KlipperScreen/KlipperScreen #1479: [FR] <title> Set a static size of the screen when printing or extruding

JuanR3D, 2024-10-11:
### Requested feature

I seems that the size of the screen get adjusted automatically when the temperature percentaje changes (i guess it does when anything changes), for example from 1% to 10% 

I'd like to set an specific size of the screen so it doesn't changes everytime.
Is this possible?

### Additional information

https://github.com/user-attachments/assets/19b3561d-dfeb-4ca2-b9ca-f65fa07892fe

https://github.com/user-attachments/assets/0c049dc4-f928-408a-9e71-8684117fc7b2

JuanR3D, 2025-02-10:
Hi @alfrix . I'm going to set an static size  for my machine by myself. Could you tell me which script should i modify in order to do this?

alfrix, 2025-02-10:
> Hi @alfrix . I'm going to set an static size  for my machine by myself. Could you tell me which script should i modify in order to do this?

Extrude and job_status panels, if you make it work for common resolutions please PR it

JuanR3D, 2025-02-11:
Okay, thanks! 
I'll share any change I do in this so you can take a look!