KlipperScreen/KlipperScreen #1481: If Klipperservice starts TFT PI-TS35 stuck in animated bootpicture

DK4YA, 2024-10-12:
### What happened?

If Klipperservice starts TFT PI-TS35 stuck in animated bootpicture. If i stopp Klipperscreen the TFT shows his animated picture and so on. But the Klippersurcafe isn´t shown. I remember after any "atp" upgrade this foult comes up.

### What did you expect to happen instead?

Maybe driverproblems.

### How to reproduce this bug?

sorry, i dont know.

### Additional information:

MKS SKIPR with PI-TS35.

### Log output

2024-10-12 15:01:37,660 [functions.py:log_start()] - --------------------KlipperScreen Log Start--------------------
2024-10-12 15:01:37,662 [functions.py:log_start()] - KlipperScreen Version: v0.4.4-27-ge1459337
2024-10-12 15:01:37,662 [functions.py:log_start()] - Python version: 3.8
2024-10-12 15:01:37,815 [config.py:get_config_file_location()] - Passed config (-c): .
2024-10-12 15:01:37,817 [config.py:__init__()] - Config path location: /home/mks/printer_data/config/KlipperScreen.conf
2024-10-12 15:01:37,820 [config.py:_include_config()] - Parsing files: ['/home/mks/KlipperScreen/config/main_menu.conf']
2024-10-12 15:01:37,841 [config.py:_include_config()] - Parsing files: ['/home/mks/KlipperScreen/config/splash_menu.conf']
2024-10-12 15:01:37,847 [config.py:_include_config()] - Parsing files: ['/home/mks/KlipperScreen/config/print_menu.conf']
2024-10-12 15:01:37,859 [config.py:_include_config()] - Parsing files: ['/home/mks/KlipperScreen/config/move_menu.conf']
2024-10-12 15:01:37,872 [config.py:log_config()] -  ===== Config File =====

=======================
2024-10-12 15:01:37,875 [config.py:__init__()] - ====== Saved Def ======

[main]
language = en

=======================
2024-10-12 15:01:37,877 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_path": "",
      "moonraker_ssl": null,
      "moonraker_api_key": ""
    }
  }
]
2024-10-12 15:01:38,046 [config.py:create_translations()] - Selected lang: en OS lang: en_HK
2024-10-12 15:01:38,046 [config.py:install_language()] - Using lang en
2024-10-12 15:01:38,071 [screen.py:__init__()] - Display for xset: :0
2024-10-12 15:01:38,073 [screen.py:__init__()] - Screen 0: 1024x768
2024-10-12 15:01:38,074 [screen.py:__init__()] - Monitors: 1 using number: 0
2024-10-12 15:01:38,074 [screen.py:__init__()] - Wayland: False Display name: :0
2024-10-12 15:01:38,078 [screen.py:__init__()] - Screen resolution: 1024x768
2024-10-12 15:01:38,337 [KlippyGtk.py:__init__()] - Font size: 25.6 (medium)
2024-10-12 15:01:38,507 [base_panel.py:battery_percentage()] - Battery information not available.
2024-10-12 15:01:38,582 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2024-10-12 15:01:38,598 [screen.py:set_screenblanking_timeout()] - Using DPMS
2024-10-12 15:01:38,610 [screen.py:initial_connection()] - Default printer: None
2024-10-12 15:01:38,669 [screen.py:connect_printer()] - Connecting to printer: Printer
2024-10-12 15:01:38,670 [screen.py:_remove_all_panels()] - Removing all panels
2024-10-12 15:01:38,719 [screen.py:_load_panel()] - Loading panel: splash_screen
2024-10-12 15:01:38,798 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2024-10-12 15:01:38,832 [KlippyWebsocket.py:connect()] - Attempting to connect
2024-10-12 15:01:38,833 [KlippyWebsocket.py:connect()] - Starting websocket thread
2024-10-12 15:01:38,850 [_logging.py:info()] - Websocket connected
2024-10-12 15:01:38,850 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2024-10-12 15:01:38,898 [screen.py:websocket_connected()] - ### websocket_connected
2024-10-12 15:01:38,960 [KlippyWebsocket.py:identify_client()] - Sending server.connection.identify
2024-10-12 15:01:38,983 [printer.py:configure_cameras()] - Cameras: []
2024-10-12 15:01:39,001 [screen.py:init_klipper()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'ready', 'components': ['secrets', 'template', 'klippy_connection', 'jsonrpc', 'internal_transport', 'application', 'websockets', 'database', 'dbus_manager', 'file_manager', 'authorization', 'klippy_apis', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'history', 'http_client', 'announcements', 'webcam', 'extensions', 'update_manager', 'octoprint_compat'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes', 'config_examples', 'docs'], 'warnings': [], 'websocket_count': 4, 'moonraker_version': 'v0.9.3-0-g71f9e67', 'missing_klippy_requirements': [], 'api_version': [1, 4, 0], 'api_version_string': '1.4.0'}
2024-10-12 15:01:39,194 [printer.py:log_counts()] - Klipper version: v0.12.0-317-gb89d55238
2024-10-12 15:01:39,195 [printer.py:log_counts()] - # Extruders: 1
2024-10-12 15:01:39,195 [printer.py:log_counts()] - # Temperature devices: 1
2024-10-12 15:01:39,195 [printer.py:log_counts()] - # Fans: 2
2024-10-12 15:01:39,196 [printer.py:log_counts()] - # Output pins: 0
2024-10-12 15:01:39,196 [printer.py:log_counts()] - # PWM tools: 0
2024-10-12 15:01:39,196 [printer.py:log_counts()] - # Leds: 0
2024-10-12 15:01:39,447 [KlippyWebsocket.py:object_subscription()] - Sending printer.objects.subscribe
2024-10-12 15:01:39,448 [files.py:set_gcodes_path()] - Gcodes path: /home/mks/printer_data/gcodes
2024-10-12 15:01:39,449 [screen.py:init_klipper()] - Printer initialized
2024-10-12 15:01:39,450 [printer.py:change_state()] - Changing state from 'disconnected' to 'ready'
2024-10-12 15:01:39,450 [printer.py:change_state()] - Adding callback for state: ready
2024-10-12 15:01:39,464 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2024-10-12 15:01:39,476 [screen.py:set_screenblanking_timeout()] - Using DPMS
2024-10-12 15:01:39,487 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
2024-10-12 15:01:39,492 [screen.py:_remove_all_panels()] - Removing all panels
2024-10-12 15:01:39,548 [screen.py:_load_panel()] - Loading panel: main_menu
2024-10-12 15:01:39,696 [main_menu.py:__init__()] - ### Making MainMenu
2024-10-12 15:01:39,757 [main_menu.py:add_device()] - Adding device: extruder
2024-10-12 15:01:39,767 [main_menu.py:add_device()] - Adding device: heater_bed
2024-10-12 15:01:39,814 [menu.py:arrangeMenuItems()] - X > gcodes
2024-10-12 15:01:39,833 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2024-10-12 15:01:39,865 [printer.py:init_temp_store()] - Temp store: ['heater_bed', 'extruder']
2024-10-12 15:01:39,886 [screen.py:init_tempstore()] - Temperature store size: 1200

alfrix, 2024-10-13:
klipperscreen seems to be starting correctly, check out this:
https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/Framebuffer/

closing since it's not a ks bug

closed 2024-10-13 by alfrix

DK4YA, 2024-10-13:
Can I find my trouble in troubleshhotig section?

[image]

alfrix, 2024-10-13:
> Can I find my trouble in troubleshhotig section? [...]

klipperscreen is definitively starting and working in the log you provided, 

if the fb is not the issue, then try to add this to klipperscreen.conf
```
[main]
use_dpms = False
```
and restart the service