KlipperScreen/KlipperScreen #1505: [BUG] <title>White screen after wake up

3D-Cat, 2024-11-15:
### What happened?

after the screen turns black and I'm trying to wake up the screen, I see only a white display. Even though you can't see anything the KlipperScreen reacts on touch. I've this issue with the last KlipperScreen update (v0.4.5-10-g43ad9fc6)

### What did you expect to happen instead?

readable screen after wake up

### How to reproduce this bug?

waking up after the screen is blacked out

### Log output

[KlipperScreen.zip](https://github.com/user-attachments/files/17774215/KlipperScreen.zip)

closed 2024-11-17 by alfrix

3D-Cat, 2024-11-17:
any idea why this happens? I've no issue when I turn off the screen saver. thanks

alfrix, 2024-11-18:
> it was working without any issues for more then 1 year, anyway I'll disable dpms. thanks

interesting, if it's really klipperscreen, you can try to revert to a previous version from ssh and test

for example something like this:

```
cd ~/klipperscreen
git reset --hard v0.4.0
sudo service KlipperScreen restart
```

some screens fake dpms making the screen black (GPIO screens for example), if the screen doesn't turn off it's pointless to continue trying to solve it

3D-Cat, 2024-11-18:
after disabling DPMS screen turns black after defined time and wake up causes no issues.
I'm not sure if DPMS was enabled or not before update but for me it's OK how it's works now. 
many thanks for your help and grate work

3D-Cat, 2024-11-18:
btw I'm using BTT Pad 7 with CM4

commit 301514f4 (2024-11-17): fix: screensaver issues when using dpms

close #1505