KlipperScreen/KlipperScreen #1510: [BUG] <title>Swap Memory Leak

JorgeSDF, 2024-11-21:
### What happened?

Hiii, Im using Klipperscreen with a Rpi Zero 2W and it works great!! BUT...when using klipperscreen for too long the swap memory fills and collapses the system. If I restart the service it goes down. Im monitoring the Ram usage and Swap memory with Htop while printing and using the touch interface. Klipperscreen will release Ram but not release the swap memory🥲
[image]

### What did you expect to happen instead?

I expect that klipperscreen will release the swap memory to prevent timer too close issues.
[image]

### How to reproduce this bug?

Use klipperscreen and scroll rapidly up and down to reproduce the bug as fast as possible, you will need a system with 512MB of RAM

### Additional information:

Im using a 7Inch Touchscreen, a Webcam at Low resolution(640x480) and a BDSensor.

### Log output

[logs-20241121-110932.zip](https://github.com/user-attachments/files/17848852/logs-20241121-110932.zip)

alfrix, 2024-11-23:
The kernel is responsible of freeing the swap pages, it will depend a bit on the system configuration, a regular application cannot 'free' swap pages, the memory

in your screenshots there is no leak in sight, you just have a very little amount of ram that is causing issues, klipperscreen  starts with around 50mb  of usage but it will grow to 150-200mb ram depending on the panels that are being loaded.

what i would recommend for systems with this low amount of memory is enabling (zram)[https://wiki.debian.org/ZRam] and disabling the swap file

```
sudo apt install zram-tools
echo -e "ALGO=zstd\nPERCENT=80" | sudo tee -a /etc/default/zramswap
sudo service zramswap reload
sudo dphys-swapfile swapoff
```

installing earlyoom will kill the app that uses the most memory and prevent printer crashes or failures if the system runs out of memory

```
sudo apt install earlyoom
```

i've done all of the above in my PI3A+ which should be almost equal to the zero2w

i've noticed that you have less available overall ram than i do, so the ram split has been tweaked at some point, my 3A uses 64mb of video memory by default, maybe you can also reduce the cpu/gpu memory split in config.txt

``` 
gpu_mem=64
```
i hope this helps you

closed 2024-11-23 by alfrix

JorgeSDF, 2024-11-23:
@alfrix Thank you very much for your help and guidance!!! Appreciate the time and effort you've taken to assist me!! I will try that out!!!