KlipperScreen/KlipperScreen #1519: [BUG] Move-Home button causes KS to freeze instead of displaying options for homing

ChristopherUC, 2024-12-02:
### What happened?

After installing Commit 4172d59 and newer, Pressing Move-Home on my display causes KS to freeze instead of displaying the various options for homing. 

Kiauh reports the last installed functioning version as "v0.4.5-8" when I choose update. 

### What did you expect to happen instead?

I expected the homing menu to load instead of KS to freeze. 

### How to reproduce this bug?

(on my printer) From the main/Home Screen, press "Move". From the "Move" screen, press "Home"

I expect this must be working for many people as it would be very noticeable. 
I have uninstalled-reinstalled KS using KIAUH (v5.1.2) several times. I have updated system packages and ensured that the venv is re-created. 

I can do the following to get it working again: 
$ cd ~/KlipperScreen/
$ git checkout f4c6801a8c562aa9f505864add76060d036b723b
https://github.com/KlipperScreen/KlipperScreen/tree/f4c6801a8c562aa9f505864add76060d036b723b

I worked commit-by-commit and the first commit where this stops working on my device can be loaded with
$ git checkout 4172d59dbe547334e3b203ad3445b039b7007915
https://github.com/KlipperScreen/KlipperScreen/tree/4172d59dbe547334e3b203ad3445b039b7007915

### Additional information:

Working Log: (version "v0.4.5-8" / commit f4c6801 / cleared and service restarted) 
~/printer_data/logs$ cat KlipperScreen.log 
2024-12-02 14:27:21,944 [functions.py:log_start()] - --------------------KlipperScreen Log Start--------------------
2024-12-02 14:27:21,945 [functions.py:log_start()] - KlipperScreen Version: v0.4.5-8-gf4c6801a
2024-12-02 14:27:21,946 [functions.py:log_start()] - Python version: 3.9
2024-12-02 14:27:22,090 [config.py:get_config_file_location()] - Passed config (-c): .
2024-12-02 14:27:22,102 [config.py:__init__()] - Config path location: /home/biqu/printer_data/config/KlipperScreen.conf
2024-12-02 14:27:22,107 [config.py:_include_config()] - Parsing files: ['/home/biqu/KlipperScreen/config/main_menu.conf']
2024-12-02 14:27:22,124 [config.py:_include_config()] - Parsing files: ['/home/biqu/KlipperScreen/config/splash_menu.conf']
2024-12-02 14:27:22,131 [config.py:_include_config()] - Parsing files: ['/home/biqu/KlipperScreen/config/print_menu.conf']
2024-12-02 14:27:22,142 [config.py:_include_config()] - Parsing files: ['/home/biqu/KlipperScreen/config/move_menu.conf']
2024-12-02 14:27:22,160 [config.py:log_config()] - ===== Config File =====
[menu __main custom]
name = Prepare
icon = klipper

[menu __main custom present]
name = Present Toolhead
icon = bed-level-t-r
method = printer.gcode.script
params = {"script":"present_toolhead"}

[menu __main custom load]
name = Load Clean
icon = arrow-down
method = printer.gcode.script
params = {"script":"load_clean"}

[menu __main custom unload]
name = Unload Clean
icon = arrow-up
method = printer.gcode.script
params = {"script":"unload_clean"}

[menu __main custom stow]
name = Stow Toolhead
icon = bed-level-b-l
method = printer.gcode.script
params = {"script":"stow_toolhead"}

[menu __main custom ready_up_pla]
name = Ready Up PLA
[... 302 lines cut ...]
$ journalctl -xe -u KlipperScreen > ~/printer_data/logs/KlipperScreen_journalctl.log
$ tail KlipperScreen_Xorg.log 
[  9946.606] (II) modeset(0): Modeline "800x600"x0.0   40.00  800 840 968 1056  600 601 605 628 +hsync +vsync (37.9 kHz e)
[  9946.606] (II) modeset(0): Modeline "1024x768"x0.0   65.00  1024 1048 1184 1344  768 771 777 806 -hsync -vsync (48.4 kHz e)
[  9946.606] (II) modeset(0): Modeline "1152x864"x0.0  108.00  1152 1216 1344 1600  864 865 868 900 +hsync +vsync (67.5 kHz e)
[  9946.606] (II) modeset(0): Modeline "1280x800"x0.0   71.00  1280 1328 1360 1440  800 803 809 823 +hsync -vsync (49.3 kHz e)
[  9946.606] (II) modeset(0): Modeline "1280x960"x0.0  108.00  1280 1376 1488 1800  960 961 964 1000 +hsync +vsync (60.0 kHz e)
[  9946.606] (II) modeset(0): Modeline "1280x1024"x0.0  108.00  1280 1328 1440 1688  1024 1025 1028 1066 +hsync +vsync (64.0 kHz e)
[  9946.606] (II) modeset(0): Modeline "1440x900"x0.0   88.75  1440 1488 1520 1600  900 903 909 926 +hsync -vsync (55.5 kHz e)
[  9946.606] (II) modeset(0): Modeline "1600x900"x60.0  119.00  1600 1696 1864 2128  900 901 904 932 -hsync +vsync (55.9 kHz e)
[  9946.606] (II) modeset(0): Modeline "1600x1200"x0.0  162.00  1600 1664 1856 2160  1200 1201 1204 1250 +hsync +vsync (75.0 kHz e)
[  9946.607] (--) modeset(0): HDMI max TMDS frequency 150000KHz

$cp /var/log/Xorg.0.log ~/printer_data/logs/KlipperScreen_Xorg.log
$ tail KlipperScreen_journalctl.log
Dec 02 14:32:10 SV08 KlipperScreen-start.sh[54261]: 20241202 14:32:10,853 [menu.py:arrangeMenuItems] - X > gcodes
Dec 02 14:32:11 SV08 KlipperScreen-start.sh[54261]: 20241202 14:32:11,013 [screen.py:attach_panel] - Current panel hierarchy: main_menu
Dec 02 14:32:11 SV08 KlipperScreen-start.sh[54261]: 20241202 14:32:11,077 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_sensor mcu_temp', 'temperature_sensor Host_temp', 'temperature_sensor Toolhead_Temp', 'extruder']
Dec 02 14:32:11 SV08 KlipperScreen-start.sh[54261]: 20241202 14:32:11,107 [screen.py:init_tempstore] - Temperature store size: 1200
Dec 02 14:32:50 SV08 KlipperScreen-start.sh[54261]: 20241202 14:32:50,988 [screen.py:_load_panel] - Loading panel: move
Dec 02 14:32:51 SV08 KlipperScreen-start.sh[54261]: 20241202 14:32:51,225 [screen.py:attach_panel] - Current panel hierarchy: main_menu > move
Dec 02 14:32:52 SV08 KlipperScreen-start.sh[54261]: 20241202 14:32:52,683 [screen.py:_load_panel] - Loading panel: menu
Dec 02 14:32:53 SV08 KlipperScreen-start.sh[54261]: 20241202 14:32:53,468 [screen.py:attach_panel] - Current panel hierarchy: main_menu > move > menu
Dec 02 14:32:54 SV08 KlipperScreen-start.sh[54261]: 20241202 14:32:54,010 [menu.py:arrangeMenuItems] - X > homing Z-Tilt
Dec 02 14:33:17 SV08 KlipperScreen-start.sh[54261]: 20241202 14:33:17,030 [base_panel.py:battery_percentage] - Battery information not available.

ChristopherUC, 2024-12-02:
please let me know if there is anything further I can test/do to assist. Apologies if there's something simple I missed that is causing this for me. I understand this is likely an edge-case issue since this is so noticeable that others would obviously have reported it weeks ago.

JuanR3D, 2024-12-03:
I can confirm this issue as well. After updating KlipperScreen, the Home button no longer works.

Initially, I thought the issue might be on my end, as I had modified move.py some time ago. When I attempted to update, Git flagged conflicts between the file in the repository and my modified version. However, after performing a complete reinstallation of KlipperScreen, I can confirm that this problem is not caused by my modifications.
[klippy.log](https://github.com/user-attachments/files/17990387/klippy.log)

I've attached my logs for reference.

Hayvosh, 2024-12-03:
I can also confirm this issue, having the exact same problem on my RPi3B+ running KlipperScreen.

igeighty, 2024-12-04:
issue exists also on klipperscreen running on a BTT cb1+manta8p also on 2 other printers running raspberry3b+btt piTFT.
walk up to any of these and hit 'move' then 'home' thats when klipperscreen freezes, v0.4.5-24-g7431ad82

JuanR3D, 2024-12-04:
I also have a CB1 running with a Manta M8P for this. Also mention that now the screen is much slower than before in general.

JuanR3D, 2024-12-04:
Hey, I have a question about this...  After updating a service (like KlipperScreen or others), how can I manually revert to a previous version and reinstall it? Any tips or best practices for handling this across different services?"

ChristopherUC, 2024-12-05:
> Hey, I have a question about this... After updating a service (like KlipperScreen or others), how can I manually revert to a previous version and reinstall it? Any tips or best practices for handling this across different services?"

@JuanR3D 
I know it was really lost in my HUGE initial bug report, but the way I rolled back to the working version was as follows. 

```
<SSH into printer>
<best practice would be to stop the KlipperScreen service before doing this, though it should work regardless>
cd ~/KlipperScreen/
git checkout f4c6801
<Restart at a minimum KlipperScreen, preferably the whole printer>
```
Important Note: Don't update KlipperScreen via Kiauh, web interface, etc, after this until/unless you're ready to go back to the current version.

ChristopherUC, 2024-12-05:
> I also have a CB1 running with a Manta M8P for this. Also mention that now the screen is much slower than before in general.

I confirm I'm also seeing this reduction in overall responsiveness on the newer version. I'm not certain how to quantify it in a data oriented way. ItItss not quite bad enough I could use a stopwatch for it, but there is a definitive delay in screen update that isn't there in prior versions on my printer as well.

JuanIgnacioMunozSanchez, 2024-12-05:
@ChristopherUC, how did you determine that you had to use that specific commit number (f4c6801)? I searched and found that this is the hash of a specific commit (https://github.com/KlipperScreen/KlipperScreen/pull/1494) so I understand it will revert the changes up to that commit. However, I’m curious—how did you identify that this particular commit was the right one?

Additionally, is there a way to revert to the machine's previous state before the update, rather than to a specific commit? I assume it would involve using a user backup, but I’d like confirmation on whether this is correct or if there's another approach.

ChristopherUC, 2024-12-05:
> @ChristopherUC, how did you determine that you had to use that specific commit number ([f4c6801](https://github.com/KlipperScreen/KlipperScreen/commit/f4c6801a8c562aa9f505864add76060d036b723b))? I [...]

Honestly? I'm not saying that I'd recommend this for everyone, but I remembered it had broken some time in late September or early October and I'd been living with it since then. I went to the commit history, pulled a hash from some time in September, and did a 'git checkout' on it. I worked forward that way one hash at a time until I found the one I confirmed the Home button 'broke' in. I also checked a couple hashes forward from that point, and verified it remained broken. Then I went back to the one I believed it was fixed in, and confirmed that was the last commit that works on my machine. 
 I did not test every hash individually between that specific point and the most current. I spent a fair bit of extra time after finding the last working hash hoping that by isolating it to a specific commit it would make for a better bug report for the KS team. 
It's probably FAR safer to use a release tag.  e.g. "git checkout tags/<tag> -b <branch>" and I would likely recommend that for most. 

> Additionally, is there a way to revert to the machine's previous state before the update, rather than to a specific commit? I assume it would involve using a user backup, but I’d like confirmation on whether this is correct or if there's another approach.

You raise an interesting question and one I'm looking into as I write this. I probably should have looked into this earlier.  I can confirm the Kiauh backup function, under the main menu option 5-Backup, exists.  It also does have an option for KlipperScreen (along with other installed components).  After testing it, my initial impression is that it appears to just create a copy of the KlipperScreen directory into ~/kiauh-backups/klipperscreen-backups. I would guess you could just copy it back following an unsuccessful update, but I haven't tried. 

I know I must have used it before as there were 2 subdirectories in the backup folder.  That said, I didn't see an easy 'restore' button. I'm off to check the documentation now. Assuming there isn't one that I've overlooked, I think I will see if I can mock something like that up for a possible PR in the future assuming the devs there are open to it and I can find the time to get it done. 

Worth noting, under kiauh option "6-Settings" there is also item "4-Enable Automatic backups before updates". I will likely turn this on going forward.  

I also personally use klipper-backup [https://klipperbackup.xyz] to push my ~/printer_data/config directory to GitHub so I have an off-machine backup of those essential files. I STRONGLY recommend using a private GitHub repo for this, NOT public, PARTICULARLY if you use Obico.  The file 'moonraker-obico.cfg' contains an API auth token that should NOT be exposed publicly.

igeighty, 2024-12-05:
thanks for the rollback tips @JuanR3D i have rollled back also and it is working as expected, screen is also much more responsive.     i might say that i am running demon klipper scripts https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified also. later tonight i will try the new klipperscreen with just a default printer.cfg and see if the DK scripts are conflicting with klipperscrren

DrumClock, 2024-12-06:
Hi, Same problem KlipperScreen with version v0.4.5-24-g7431ad82

JuanIgnacioMunozSanchez, 2024-12-06:
Hi again @ChristopherUC .

Thanks a lot for the advice! I'll take a look in the backup options of kiauh, but I have another question about this.
When I update klippescreen from update manager of mainsail, this is done through kiauh or how? I mean,,, Will this functionality avaliable in this case?

And yes! I have a private git repo that contains all my cfg files :)

Thanks again

JuanIgnacioMunozSanchez, 2024-12-06:
Also, I have a question about using a GitHub repository to manage my machine setup. I'm not sure if this is the right place to ask—if not, please let me know where I should post this, and I’ll remove this message.

I’d like to replicate the same configuration across multiple machines as quickly and automatically as possible. This includes not just the config files (which is simple—I upload them and adjust the control board IDs), but also everything else: Mainsail configurations, startup logos, installed dependencies, etc.

Currently, I follow a guide I wrote myself to avoid missing steps, but it feels slow. My process involves flashing the Mainsail image (with Klipper services pre-installed) onto the machine, then installing all the required dependencies and services manually.

I’m considering using a Git repository for this. I assume it should include all the machine’s information to replicate, but I’m not sure how to proceed. Does anyone have suggestions?

Thanks in advance

commit d0faec9c (2024-12-06): Revert "feat: Menu active status (#1482)"

This reverts commit 4172d59dbe547334e3b203ad3445b039b7007915.

due to a bug that needs to be investigated

closes #1519
closed 2024-12-06 by alfrix