KlipperScreen/KlipperScreen #1531: [FR] RESPOND Behavior

Sineos, 2025-01-06:
### Requested feature

Align RESPOND behavior with Mainsail / Fluidd

### Is your feature request related to a problem?

I would like KlipperScreen to show the same behavior regarding the `RESPOND` command as Mainsail or Fluidd.

1. `RESPOND` shows nothing when using a `PREFIX`.
2. When using macros to format `RESPOND` (see [this example](https://klipper.discourse.group/t/extend-respond-command-with-some-color-options/4158)), the formatting is not rendered but passed literally.

In any case, thanks for this fine product.

### Describe alternatives you have considered

n/a

### Additional information

n/a

alfrix, 2025-01-12:
1. The respond messages of type command (prefixed by `//`) and with other prefixes will appear on the console panel

popups are identified by their prefixes, and there is no way around that, moonraker will send a websocket message like this [more info](https://moonraker.readthedocs.io/en/latest/web_api/#gcode-response):
```
{
    "jsonrpc": "2.0",
    "method": "notify_gcode_response",
    "params": ["response message"]
}
```
the key is that notify_gcode_response is not unique to RESPOND is a generic interface used by all the commands, so the only way you can pinpoint a popup is by the prefix of the message

2. That seems like a hack around the fact that those interfaces run on a browser that can parse HTML, to make that work i would have to add an HTML parser and run every message thorough it, and that will likely would likely introduce significant processing overhead.

In KS there are very specific css classes of which there are only 3 that format the popups with different colors:
green level for `echo:` (`RESPOND TYPE=echo` or without type )
red level for `!!` (`RESPOND TYPE=error` )
yellow level is only used internally since klipper doesn't have a warn level in the respond types

I hope that clears up the situation

Sineos, 2025-01-12:
Thanks a lot for the response. Appreciated.
Actually I was expecting something along these lines and it perfectly makes sense.

It was just the vague hope that there might be an "easy" way to align the behavior between the products.

> 2\. That seems like a hack around the fact that those interfaces run on a browser that can parse HTML, to make that work i would have to add an HTML parser and run every message thorough it, and that will likely would likely introduce significant processing overhead.

Of course, like a lot of macros as well. But at least it looks nice 😉

closed 2025-01-12 by Sineos