KlipperScreen/KlipperScreen #1544: [BUG] DPMS toggle no longer working with TFT50 screen

Dark360, 2025-02-09:
### What happened?

My TFT50 screen is now showing multiple colours regardless of the DPMS toggle in settings.

### What did you expect to happen instead?

The TFT50 screen to go black.

### How to reproduce this bug?

Update to the latest version of Klipperscreen
v0.4.5-42-g57b1749d

### Additional information:

Raspberry Pi 4
BTT TFT50

### Log output

[KlipperScreen.log](https://github.com/user-attachments/files/18723630/KlipperScreen.log)

Dark360, 2025-02-09:
Issue is not present in v0.4.5-39-gacae6a29

[KlipperScreen_gacae6a29.log](https://github.com/user-attachments/files/18723672/KlipperScreen_gacae6a29.log)

CBoismenu, 2025-02-10:
I'm having the same issue.

Raspberry Pi 4 and BTT TFT70.

alfrix, 2025-02-12:
@Dark360 i cannot open your original log, please re-attach it.

the secondary log shows dpms off, it aligns with my expectations that Pi TFT doesn't properly support DPMS

alfrix, 2025-02-12:
please test the screen by running this from ssh:

disable blanking (i suspect this is the issue):
`xset -display :0 s off`

force screen off via dpms:
`xset -display :0 dpms force off`

force screen on via dpms:
`xset -display :0 dpms force on`

set times in seconds for dpms (maybe one of this works):
`xset -display :0 dpms 15 30 45`

times are in seconds, you should wait 15 seconds for standby, 30 for suspend and 45  for off (feel free to reduce them) the idea is to see if it enters said mode and you can wake it up with touch afterwards.

ks sets the suspend time only, but it remains possible that we need to set the off time too

please let me know if you test it how it went.

Dark360, 2025-02-15:
Here is the Log again [KlipperScreen.log](https://github.com/user-attachments/files/18811516/KlipperScreen.log)

Do you want me to test these things on the current branch or on v0.4.5-39-gacae6a29?

Dark360, 2025-02-15:
> please test the screen by running this from ssh:
>
> disable blanking (i suspect this is the issue): `xset -display :0 s off`
>
> force screen off via dpms: `xset -display :0 dpms force off`
>
> force screen [...]

`xset -display :0 s off` = Still shows colours

`xset -display :0 dpms force off` = Turns the screen off as expected, but still shows colours

`xset -display :0 dpms force on` = Turns the screen on as expected

`xset -display :0 dpms 15 30 45` = Still shows colours

I was able to revert original functionailty using `xset -display :0 -dpms`, so it appears that the DPMS option is not disabling DPMS on the screen.

andyvortex, 2025-02-16:
Yeah mines doing same to

alfrix, 2025-02-16:
> Screen didn't show boot log

@Fuchs-Fox if this happens then there is a problem with your hw, either cable, power-supply or screen, so unrelated to this issue.

>xset -display :0 s off = Still shows colours

so blanking and colors are unrelated

>xset -display :0 dpms force off = Turns the screen off as expected and enables DPMS, but still shows colours
>xset -display :0 dpms force on = Turns the screen on as expected, and enables DPMS as expected

if the display turns off and on means that display understand dpms, and can be used

>xset -display :0 dpms 15 30 45 = Still shows colours

@Dark360 shows colors after 15 30 or 45 seconds?

>I was able to revert original functionailty using xset -display :0 -dpms, so it appears that the DPMS option is not disabling DPMS on the screen.

you are right the new code does not explicitly disable dpms, this should fix the colors

the real shame on this is that if the screen does turn On/Off with the `force` option it means it should be capable of turning off the backlight, the question would be how to tell the driver to not do that color thing

try:
```
xset s noblank
xset -display :0 dpms force off
```

andyvortex, 2025-02-16:
Hardware issue unlikely I can put a card in with none updated and doesn't do it.  
It's a firmware thing 100$

andyvortex, 2025-02-16:
[image]

[image]

[image]

[image]
4 mins no flashing

andyvortex, 2025-02-16:
Not hardware

alfrix, 2025-02-16:
@andyvortex the response was for Fuchs-Fox, if your screen doesn't turn on at all not even during boot, it's definitevely a hw problem, you are supposed to see the console starting unless you explicitly hide it in the OS

andyvortex, 2025-02-16:
Sorry just thought I'd weigh in.
Let it not be said your work is awesome man it wasn't a criticism in anyway.

All you guys do is appreciated period.

I kinda like hassles give me something to do 😂

Dark360, 2025-02-16:
> [@Fuchs-Fox](https://github.com/Fuchs-Fox) if this happens then there is a problem with your hw, either cable, power-supply or screen, so unrelated to this issue.
>
>
> so blanking and colors are [...]

So regarding the whole DPMS not turning off the Backlight on the PI TFT50.
This is a known issue of the PI TFT 50 v1.0-v2.0 as they have potentiometer to control brightness, so we will never be able to control it via firmware/software.
They removed the potentiometer in the PI TFT50 v2.1, so we can now control the backlight:
[https://github.com/bigtreetech/BIGTREETECH-TouchScreenHardware/issues/14](url)

I have a V2.0, so I have accepted that.

Just incase it helps.

With DPMS turned on in Config.

Screen On:

Screen Saver:
  prefer blanking:  **yes**    allow exposures:  yes
  timeout:  0    cycle:  600
DPMS (Energy Star):
  Standby: 0    Suspend: 60    Off: 0
  DPMS is Enabled
  **Monitor is On**

Screen Showing Colours:

Screen Saver:
  prefer blanking:  **yes**    allow exposures:  yes
  timeout:  0    cycle:  600
DPMS (Energy Star):
  Standby: 0    Suspend: 60    Off: 0
  DPMS is Enabled
  **Monitor is in Suspend**

With DPMS turned off in Config.

Screen On:

Screen Saver:
  prefer blanking:  **yes**    allow exposures:  yes
  timeout:  600    cycle:  600
DPMS (Energy Star):
  Standby: 600    Suspend: 600    Off: 600
  DPMS is Enabled
  **Monitor is On**

Screen Showing Black Screen/Backlight:

Screen Saver:
  prefer blanking:  **yes**    allow exposures:  yes
  timeout:  600    cycle:  600
DPMS (Energy Star):
  Standby: 600    Suspend: 600    Off: 600
  DPMS is Enabled
  **Monitor is On**

`xset s noblank`
xset:  unable to open display ""

`xset -display :0 s noblank
`xset -display :0 dpms force off`

Screen Showing Colours:

Screen Saver:
  prefer blanking:  **no**    allow exposures:  yes
  timeout:  600    cycle:  600
DPMS (Energy Star):
  Standby: 600    Suspend: 600    Off: 600
  DPMS is Enabled
  **Monitor is Off**

alfrix, 2025-02-16:
i suppose that the color flashing is like the No-Signal banner, ok then i'll just merge the fix for explicit disable when dpms is turned off

closed 2025-02-16 by alfrix

Dark360, 2025-02-17:
> i suppose that the color flashing is like the No-Signal banner, ok then i'll just merge the fix for explicit disable when dpms is turned off

I have updated Klipperscreen to v0.4.5-49-g38348720 and it is still going to Colours whether DPMS is toggled On or Off.

I would prefer that the it goes to a Black screen, not the Colours when DPMS is toggled off.

`xset -display :0 -dpms` is the command to disable DPMS, which is what I assume the DPMS toggle is meant to do?

Dark360, 2025-02-17:
> I have updated Klipperscreen to v0.4.5-49-g38348720 and it is still going to Colours whether DPMS is toggled On or Off.
>
> I would prefer that the it goes to a Black screen, not the Colours when DPMS [...]

OK, so this wierd.

If I toggle DPMS off, then restart KlipperScreen it goes back to showing a black screen with backlight.

However after sometime (I think 10 Mins/600 Seconds) it reverts to the Colours.

I think this has something to do with the below:

After toggling DPMS Off

Screen Saver:
  prefer blanking:  no    allow exposures:  yes
  timeout:  0    cycle:  600
DPMS (Energy Star):
  **Standby: 600    Suspend: 600    Off: 600**
  DPMS is Enabled
  Monitor is On

After toggle DPMS On

Screen Saver:
  prefer blanking:  no    allow exposures:  yes
  timeout:  0    cycle:  600
DPMS (Energy Star):
  **Standby: 0    Suspend: 60    Off: 0**
  DPMS is Enabled
  Monitor is On

Toggling DPMS is changing the values back to 600 and is not disabling DPMS as intended.

Running this `xset -display :0 dpms 0 75 0` after KlipperScreen restarts confirms this as it is going to Colours with DPMS toggled off 15 Seconds after it enters the screensaver.

Dark360, 2025-02-17:
@alfrix I have just updated to v0.4.5-50-ge71fe755 and the same issue is still present.

After 60 seconds the screen goes blank (as expected) and then after 600 seconds it goes to Colours.

alfrix, 2025-02-17:
> [@alfrix](https://github.com/alfrix) I have just updated to v0.4.5-50-ge71fe755 and the same issue is still present.
> 
> After 60 seconds the screen goes blank (as expected) and then after 600 seconds it goes to Colours.
> 
> If need any debug logs or anything let me and I will get them for you.

turn dpms off in the settings and reboot to start fresh, do not change anything and wait

Dark360, 2025-02-17:
> turn dpms off in the settings and reboot to start fresh, do not change anything and wait

Reboot the Pi or the KlipperScreen service?

I did the later.

andyvortex, 2025-02-17:
[image]
Just tired newest version. Issue still there for me too
On this build pictured no issue

Dark360, 2025-02-17:
> Reboot the Pi or the KlipperScreen service?
>
> I did the later.

Reooted the Pi and same issue. Black after 60 seconds, then colours after 600.

alfrix, 2025-02-19:
> Rebooted the Pi and same issue. Black after 60 seconds, then colours after 600.

i don't have this screen and it's impossible for me to test this, someone with the screen needs to find out the combination of settings with xset that makes the screen not do that

Dark360, 2025-02-19:
@alfrix The issue is however it used to work and now it doesn't.

Why did you change it?

My assumption would be that `xset -dpms` works as it disables DPMS, which is what I assume the DPMS toggle function is meant to do but I might be wrong as I don't know how your code works.

I appreciate the work you are doing and the function KlipperScreen provides to my 3D Printer.

I am happy to help in testing any combination of commands.

Dark360, 2025-02-20:
@alfrix `xset -display :0 dpms 0 0 0` does prevent the screensaver from coming on, which your lastest commit has added, but it is not being applied apparently.

With DPMS toggled Off (Screensaver)

Screen Saver:
  prefer blanking:  no    allow exposures:  yes
DPMS (Energy Star):
  **Standby: 600    Suspend: 600    Off: 600**
  DPMS is Enabled
  Monitor is On

With DPMS toggled Off and `xset -display :0 dpms 0 0 0` applied (No Screensaver)

Screen Saver:
  prefer blanking:  no    allow exposures:  yes
DPMS (Energy Star):
  **Standby: 0    Suspend: 0    Off: 0**
  DPMS is Enabled
  Monitor is On

Let me know if you anymore info.

sammyke007, 2025-09-28:
Is this fixed? I'm at the point of buying a Pi TFT50 v2.1 and I would like the display to go to sleep when not using it...

PR #1546 (2025-02-16): Fix Blanking
needs testing but should fix #1544