KlipperScreen/KlipperScreen #1547: [BUG] <title>Number input box does not close after submitting temperature setting.

Bifsie, 2025-02-13:
### What happened?

When clicking on the temp of either the hotend or the bed heater, it brings up an input box to enter the desired temperature. After typing in the desired number and clicking the check mark, the box remains open. This is not common behavior. The first time I did it, I thought the system locked up. 

### What did you expect to happen instead?

I expected the box to close once the temperature setting was accepted.

### How to reproduce this bug?

Click the temperature of the hotend or bed heater. Type in a number and click the check mark.

### Additional information:

This just is not common behavior. Please fix this.

### Log output

There is no log associated with this issue.

alfrix, 2025-02-16:
i would not classify this as a bug, but what you propose does make sense

Bifsie, 2025-02-16:
I was not sure this was the intended flow, so I reported it as a bug. Thank you for updating.

PR #1568 (2025-03-25): refactor: close keypad/numpad when setting a temperature
this ux change proposed makes sense to me and should close #1547
closed 2025-03-25 by alfrix