KlipperScreen/KlipperScreen #1548: Automatic start at boot not working with Wayland

rcloran, 2025-02-15:
On reboot, KlipperScreen does not successfully start until I log in by SSH. `syslog` reports:

```
seatd[640]: 00:00:13.031 [INFO] [seatd/server.c:145] New client connected (pid: 1028, uid: 1000, gid: 1003)
seatd[640]: 00:00:13.031 [INFO] [seatd/seat.c:170] Added client 1 to seat0
seatd[640]: 00:00:13.031 [INFO] [seatd/seat.c:480] Opened client 1 on seat0
KlipperScreen-start.sh[1028]: 00:00:10.010 [ERROR] [backend/backend.c:210] Found 0 GPUs, cannot create backend
KlipperScreen-start.sh[1028]: 00:00:10.010 [ERROR] [backend/backend.c:376] Failed to open any DRM device
KlipperScreen-start.sh[1028]: 00:00:10.011 [ERROR] [../cage.c:298] Unable to create the wlroots backend
seatd[640]: 00:00:23.042 [INFO] [seatd/seat.c:418] No clients on seat0 to activate
seatd[640]: 00:00:23.042 [INFO] [seatd/seat.c:524] Closed client 1 on seat0
seatd[640]: 00:00:23.042 [INFO] [seatd/seat.c:192] Removed client 1 from seat0
seatd[640]: 00:00:23.042 [INFO] [seatd/client.c:471] Client disconnected
systemd[1]: KlipperScreen.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: KlipperScreen.service: Failed with result 'exit-code'.
systemd[1]: KlipperScreen.service: Scheduled restart job, restart counter is at 2.
```

When I ssh in, /var/run/user/1000/wayland-0 is created by my SSH session, which then allows the KlipperScreen service to start successfully.

The fix for me is to to add a `PAMName` line to the `[Service]` block of the unit file:

```
diff --git a/scripts/KlipperScreen.service b/scripts/KlipperScreen.service
index bf209488..c9a30a5a 100644
--- a/scripts/KlipperScreen.service
+++ b/scripts/KlipperScreen.service
@@ -25,6 +25,8 @@ ExecStart="KS_DIR/scripts/KlipperScreen-start.sh"
 # 'who'. This is needed since we replace (a)getty.
 UtmpIdentifier=tty7
 UtmpMode=user
+# Create a PAM session to run pam-systemd, which creates /var/run/$UID
+PAMName=%u
 # A virtual terminal is needed.
 TTYPath=/dev/tty7
 TTYReset=yes
```

I've tested this change when running under X, and it also works fine -- the only difference I can tell is that a systemd login session is created:

```
❯ sudo loginctl
SESSION  UID USER    SEAT  TTY  STATE  IDLE SINCE
      2 1000 russell -     -    active no   -
     c1 1000 russell seat0 tty7 online no   -
```

(second entry, on tty7).

I'm a bit hesitant to put up a pull request to just add this, as it seems suspicious to me that nobody else has run into this. Maybe not many people run on Wayland? I'm also wary that the particular change I've suggested will change how the X config works, and I don't know systemd well enough to know if it might have any knock-on impacts. I guess it could be another thing that the install script substitutes during Wayland install?

In case it makes any difference, I'm running Ubuntu Server 24.04.

alfrix, 2025-02-16:
>I'm a bit hesitant to put up a pull request to just add this, as it seems suspicious to me that nobody else has run into this. Maybe not many people run on Wayland?

probably accurate, because is not default and when you see EXPERIMENTAL it disuades most people

in the install i've wrtitten `may need autologin` because i've used the `raspi-config` script to enable autologin to make it work

i think i've tried adding PAMName but i don't remember, it was a while ago.

the main issue is that cage doesn't support DPMS or at least it didn't at the time i made the implementation.

>  I don't know systemd well enough to know if it might have any knock-on impacts

same, make a pr if you want, a service file change will not affect existing installs only new ones.

rcloran, 2025-02-16:
> when you see EXPERIMENTAL it disuades most people
> 
> in the install i've wrtitten `may need autologin` because i've used the `raspi-config` script to enable autologin to make it work

I had initially installed with X because of that message, but decided to try Wayland to see if it would help my very slow screen (it does). Which meant I didn't really read the message on the second go around ... but I'm not sure I would've understood what autologin meant without any other context anyways.

> make a pr if you want, a service file change will not affect existing installs only new ones.

Will do.

PR #1551 (2025-02-16): fix: Start PAM session in service to allow Wayland start at boot
https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html#PAMName=

Fixes #1548
commit 38348720 (2025-02-16): fix: Start PAM session in service to allow Wayland start at boot (#1551)

https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html#PAMName=

Fixes #1548
closed 2025-02-16 by alfrix