KlipperScreen/KlipperScreen #1559: Axis coordinates permanently available on screen

RemyBaligant, 2025-03-06:
### Requested feature

Hi,

I tried different methods to show up x, y , z axis coordinates on Klipper screen without success.

My goal is to use Klipper and Klipper screen to control a z axis of a galvo fiber laser(with position memory, z homing on engraving surface with bltouch).
I will make custom menu for this application, everything is available for my application except coordinates display on screen.
For 3d printing research and development it could be a great option too.

Thanks in advance.

Best regards 
Remy

### Describe alternatives you have considered

Custom gcode

alfrix, 2025-03-09:
take a look at https://github.com/KlipperScreen/KlipperScreen/blob/master/panels/move.py
the labels are named `pos_x` `pos_y` `pos_z`

RemyBaligant, 2025-03-11:
Thanks alfrix, i understand how to use it elsewhere :) 
I close the issue. Best regards

closed 2025-03-11 by RemyBaligant