KlipperScreen/KlipperScreen #1560: [BUG] Screw tilts calculation behaviour

Escrich, 2025-03-06:
### What happened?

Hi after be in contact with Klipper 's people, they finally sent to me because a of mistake in Klipperscreen's screw tilt calculation, after having the conversation that you could see on this link:

https://klipper.discourse.group/t/screws-tilt-calculate-software-issue/22327

I finally made a Tik Tok video, showing the problem, this is the link to the aforementioned video:

https://www.tiktok.com/@josemescrich/video/7478288472009116950

_hope this help to be understanding in what I'm trying to explain.

Regards:
Escrich

### What did you expect to happen instead?

Refresh the measured value

### How to reproduce this bug?

try to make Screw tilts calculation, tap over one of the readings, and you'll see how the system take a new measure, but this new measure is not updated at all

### Additional information:

Any printer using Klipper, Klipperscreen and screws and wheels for bed leveling

### Log output

You will not see anything on any log referring this

Escrich, 2025-03-06:
I think this Klippy log will not add anything to the problem, but anyway here you have it

[klippy.log](https://github.com/user-attachments/files/19116250/klippy.log)

alfrix, 2025-03-07:
Changing a screw will affect all the others, you always have to remeasure all of them, you are supposed to do all the adjustments told and remeasure, it was explained on klipper docs. When they say repeat, it's the whole process.

>Repeat the process several times until you get a good level bed - normally when all adjustments are below 6 minutes.

https://www.klipper3d.org/Manual_Level.html#adjusting-bed-leveling-screws-using-the-bed-probe

Clicking a corner button will not remeasure that screw, it's clearly stated in the docs:

> The corner buttons in KlipperScreen place the nozzle above the screw to do the ["paper test"](https://www.klipper3d.org/Bed_Level.html?h=paper#the-paper-test).

https://klipperscreen.github.io/KlipperScreen/Panels/Screws/

closed 2025-03-07 by alfrix