KlipperScreen/KlipperScreen #1566: [BUG] `screw_rotation` does not affect Move menu

EmperorArthur, 2025-03-20:
### What happened?

Setting the `screw_rotation` configuration parameter correctly adjusts the Bed Mesh panel.  However, the buttons on the Move panel do not reflect the rotation.

Similarly, adjusting "Invert X" or "Invert Y" from the Move panel affect Bed Mesh, but not the Move panel. This is in contrast to "Invert Z" which **does** affect the Move panel buttons.

### What did you expect to happen instead?

Buttons on the Move panel will be rotated by `screw_rotation`.

### How to reproduce this bug?

1. Use the following configuration section:
```
[printer Printer]
screw_rotation: 90
```
2. Observe Bed Mesh panel reflects the change.
3. Observe Move panel does not reflect the change.

### Additional information:

This is from a Magneto X printer.
The 0,0 point is in the rear left.  This is also where the endstops are located.

### Log output

2025-03-19 23:27:50,626 [functions.py:log_start()] - --------------------KlipperScreen Log Start--------------------
2025-03-19 23:27:50,626 [functions.py:log_start()] - KlipperScreen Version: v0.4.5-53-g4395424f
2025-03-19 23:27:50,627 [functions.py:log_start()] - Python version: 3.11
2025-03-19 23:27:50,812 [config.py:get_config_file_location()] - Passed config (-c): .
2025-03-19 23:27:50,814 [config.py:__init__()] - Config path location: /home/pi/printer_data/config/KlipperScreen.conf
2025-03-19 23:27:50,819 [config.py:_include_config()] - Parsing files: ['/home/pi/KlipperScreen/config/main_menu.conf']
2025-03-19 23:27:50,842 [config.py:_include_config()] - Parsing files: ['/home/pi/KlipperScreen/config/splash_menu.conf']
2025-03-19 23:27:50,850 [config.py:_include_config()] - Parsing files: ['/home/pi/KlipperScreen/config/print_menu.conf']
2025-03-19 23:27:50,862 [config.py:_include_config()] - Parsing files: ['/home/pi/KlipperScreen/config/move_menu.conf']
2025-03-19 23:27:50,878 [config.py:log_config()] - ===== Config File =====
[main]

[preheat PLA]
bed = 60
extruder = 175

[preheat PETG]
bed = 70
extruder = 175

[printer Printer]
screw_rotation = 90
extrude_distances = 5, 10, 25, 100
extrude_speeds = 1, 5, 25, 100
=======================
2025-03-19 23:27:50,885 [config.py:__init__()] - ====== Saved Def ======

[main]
screen_blanking = 900
show_heater_power = True
invert_z = True
move_speed_xy = 1500
print_sort_dir = date_desc
language = en

[displayed_macros Printer]
line_purge = False
t0 = False
t1 = False
t2 = False
print_end = False
print_start = False

[graph Printer]
temperature_fan pi = False

=======================
2025-03-19 23:27:50,886 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_path": "",
      "moonraker_ssl": null,
      "moonraker_api_key": ""
    }
  }
]
2025-03-19 23:27:51,091 [config.py:create_translations()] - Selected lang: en OS lang: en_US
2025-03-19 23:27:51,092 [config.py:install_language()] - Using lang en
2025-03-19 23:27:51,122 [screen.py:__init__()] - Display for xset: :0
2025-03-19 23:27:51,126 [screen.py:__init__()] - Screen 0: 1024x600
2025-03-19 23:27:51,126 [screen.py:__init__()] - Monitors: 1 using number: 0
2025-03-19 23:27:51,127 [screen.py:__init__()] - Wayland: False Display name: :0
2025-03-19 23:27:51,131 [screen.py:__init__()] - Screen resolution: 1024x600
2025-03-19 23:27:51,484 [KlippyGtk.py:__init__()] - Font size: 22.2 (medium)
2025-03-19 23:27:51,745 [base_panel.py:battery_percentage()] - Battery information not available.
2025-03-19 23:27:51,937 [screen.py:set_dpms_timeout()] - DPMS on :0 set to: 900
2025-03-19 23:27:51,938 [screen.py:set_screenblanking_timeout()] - Blanking timeout: 900 DPMS:True
2025-03-19 23:27:51,940 [screen.py:initial_connection()] - Default printer: None
2025-03-19 23:27:52,050 [screen.py:connect_printer()] - Connecting to printer: Printer
2025-03-19 23:27:52,050 [screen.py:_remove_all_panels()] - Removing all panels
2025-03-19 23:27:52,147 [screen.py:_load_panel()] - Loading panel: splash_screen
2025-03-19 23:27:52,249 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2025-03-19 23:27:52,304 [KlippyWebsocket.py:connect()] - Attempting to connect
2025-03-19 23:27:52,305 [KlippyWebsocket.py:connect()] - Starting websocket thread
2025-03-19 23:27:52,320 [_logging.py:info()] - Websocket connected
2025-03-19 23:27:52,321 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2025-03-19 23:27:52,539 [screen.py:websocket_connected()] - ### websocket_connected
2025-03-19 23:27:52,598 [KlippyWebsocket.py:identify_client()] - Sending server.connection.identify
2025-03-19 23:27:52,621 [printer.py:configure_cameras()] - Cameras: [{'name': 'Redacted', 'enabled': True, 'icon': 'M12,2A7,7 0 0,1 19,9A7,7 0 0,1 12,16A7,7 0 0,1 5,9A7,7 0 0,1 12,2M12,4A5,5 0 0,0 7,9A5,5 0 0,0 12,14A5,5 0 0,0 17,9A5,5 0 0,0 12,4M12,6A3,3 0 0,1 15,9A3,3 0 0,1 12,12A3,3 0 0,1 9,9A3,3 0 0,1 12,6M6,22A2,2 0 0,1 4,20C4,19.62 4.1,19.27 4.29,18.97L6.11,15.81C7.69,17.17 9.75,18 12,18C14.25,18 16.31,17.17 17.89,15.81L19.71,18.97C19.9,19.27 20,19.62 20,20A2,2 0 0,1 18,22H6Z', 'aspect_ratio': '4:3', 'target_fps': 15, 'target_fps_idle': 5, 'location': 'printer', 'service': 'mjpegstreamer', 'stream_url': '/webcam/?action=stream', 'snapshot_url': '/webcam/?action=snapshot', 'flip_horizontal': False, 'flip_vertical': False, 'rotation': 0, 'source': 'database', 'extra_data': {}, 'uid': '78650971-0a77-47e2-bbea-b10a6ed025e1'}]
2025-03-19 23:27:52,640 [screen.py:init_klipper()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'ready', 'components': ['secrets', 'template', 'klippy_connection', 'jsonrpc', 'internal_transport', 'application', 'websockets', 'database', 'dbus_manager', 'file_manager', 'authorization', 'klippy_apis', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'history', 'http_client', 'announcements', 'webcam', 'extensions', 'octoprint_compat', 'update_manager', 'analysis'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes', 'config_examples', 'docs'], 'warnings': [], 'websocket_count': 2, 'moonraker_version': 'v0.9.3-64-g4889a2dd', 'missing_klippy_requirements': [], 'api_version': [1, 5, 0], 'api_version_string': '1.5.0'}
2025-03-19 23:27:52,797 [printer.py:log_counts()] - Klipper version: v0.12.0-443-g4113cde84-dirty
2025-03-19 23:27:52,798 [printer.py:log_counts()] - # Extruders: 3
2025-03-19 23:27:52,798 [printer.py:log_counts()] - # Temperature devices: 1
2025-03-19 23:27:52,798 [printer.py:log_counts()] - # Fans: 3
2025-03-19 23:27:52,798 [printer.py:log_counts()] - # Output pins: 0
2025-03-19 23:27:52,798 [printer.py:log_counts()] - # PWM tools: 0
2025-03-19 23:27:52,799 [printer.py:log_counts()] - # Leds: 1
2025-03-19 23:27:53,062 [KlippyWebsocket.py:object_subscription()] - Sending printer.objects.subscribe
2025-03-19 23:27:53,065 [files.py:set_gcodes_path()] - Gcodes path: /home/pi/printer_data/gcodes
2025-03-19 23:27:53,066 [screen.py:init_klipper()] - Printer initialized
2025-03-19 23:27:53,067 [printer.py:change_state()] - Changing state from 'disconnected' to 'ready'
2025-03-19 23:27:53,068 [printer.py:change_state()] - Adding callback for state: ready
2025-03-19 23:27:53,112 [screen.py:set_dpms_timeout()] - DPMS on :0 set to: 900
2025-03-19 23:27:53,113 [screen.py:set_screenblanking_timeout()] - Blanking timeout: 900 DPMS:True
2025-03-19 23:27:53,113 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
2025-03-19 23:27:53,117 [screen.py:_remove_all_panels()] - Removing all panels
2025-03-19 23:27:53,222 [screen.py:_load_panel()] - Loading panel: main_menu
2025-03-19 23:27:53,530 [main_menu.py:__init__()] - ### Making MainMenu
2025-03-19 23:27:53,556 [main_menu.py:add_device()] - Adding device: extruder
2025-03-19 23:27:53,571 [main_menu.py:add_device()] - Adding device: heater_bed
2025-03-19 23:27:53,585 [main_menu.py:add_device()] - Adding device: temperature_fan _pi
2025-03-19 23:27:53,622 [menu.py:arrangeMenuItems()] - X > gcodes
2025-03-19 23:27:53,665 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2025-03-19 23:27:53,727 [printer.py:init_temp_store()] - Temp store: ['heater_bed', 'temperature_fan _pi', 'extruder']
2025-03-19 23:27:53,748 [screen.py:init_tempstore()] - Temperature store size: 1200

alfrix, 2025-03-21:
Rotating the move buttons 90º degrees it would mean that  X and Y would have to be swapped, and that is not what you want

> Similarly, adjusting "Invert X" or "Invert Y" from the Move panel affect Bed Mesh, but not the Move panel. This is in contrast to "Invert Z" which does affect the Move panel buttons.

Invert X and Y does affect the function of the button no the label itself

> The 0,0 point is in the rear left. This is also where the endstops are located.

this is objectively a bad configuration of the printer, the 0 point is by convention the front left, you should configure the endstop position of the Y axis as the maximum Y position, klipper will home in positive direction, leaving the 0 point in the standard location

EmperorArthur, 2025-03-23:
>  that is not what you want

X and Y being swapped **is*, what I want, and expected.  At least for those buttons.

This printer's kinematics are rotated 90 degrees.  +X is towards the doors, and +Y is to the right. Imagine if a regular printer was turned 90 degrees clockwise, with the screen mounted on the side facing out.

This is what the bed mesh display shows, and why I expected the X and Y buttons to also be rotated.

> Invert X and Y does affect the function of the button no the label itself

This is inconsistent with how Invert Z functions.  That affects the labels as well.

> this is objectively a bad configuration of the printer, the 0 point is by convention the front left, you should configure the endstop position of the Y axis as the maximum Y position, klipper will home in positive direction, leaving the 0 point in the standard location

I am aware.  Unfortunately, this is the default configuration for the printer and the default OrcaSlicer profile would need serious modifications.  Especially as related to the bed image and keep out zones.

I just consider it as a standard printer with the doors and screen on the side.

closed 2026-01-28 by alfrix