KlipperScreen/KlipperScreen #1569: [FR] Add customization options for non-interative KlipperScreen install

dozed-dev, 2025-03-23:
### Requested feature

Allow choosing install options non-interactively (via script parameters or environment variables, for example)

### Is your feature request related to a problem?

I was writing a module for CustoPiZer to install KlipperScreen into a prebuilt image. However, the installation fails at restarting NetworkManager, because systemd service commands don't work in this chroot environment (see [this](https://github.com/dozed-dev/MainsailOS-dev/actions/runs/14022825304/job/39256998618#step:13:5003) workflow run).
I'd like to skip the NetworkManager step, but as it is right now, KlipperScreen install script doesn't allow for customization of install in a non-interactive environment, and the default options are selected automatically.

### Describe alternatives you have considered

I could work around that, e.g. copy and modify the install script, or try to make the systemctl commands work.
Or, KlipperScreen install script could just not run networkmanager installation step by default, or not restart the network at the end :)
But this will impact other users, and a more general solution (non-interactive customization of install options) would be nice

### Additional information

I could try to make a PR for this feature if you'd be willing to merge it

alfrix, 2025-03-25:
this is should be already possible, you need to define the variables for the questions beforehand
for example:
```sh
BACKEND="X" SERVICE="Y" START=1 NETWORK="N" ./install_script.sh
```

dozed-dev, 2025-03-27:
you're right, it already works, thank you!
maybe this should be added to the docs?

alfrix, 2025-03-31:
> you're right, it already works, thank you! maybe this should be added to the docs?

sure, you can add it if you want, maybe in a collapsed section like "advanced: unattended install" or something like that